ZipCPU/wb2axip #7: demofull incorrect read address generation in some scenarios.

MegabytePhreak, 2019-06-10:
The  read address generation incorrectly depends on the present instead of latched values of ARBURST and ARLEN if the present value of either is non-zero. I think it's a typo in the ternary operator conditions where get_next_rd_addr is instantiated.

> 	axi_addr #(.AW(AW), .DW(DW))
>		get_next_rd_addr((S_AXI_ARREADY ? S_AXI_ARADDR : raddr),
>				(S_AXI_ARREADY  ? S_AXI_ARSIZE : rsize),
>				(S_AXI_ARBURST  ? S_AXI_ARBURST: rburst), <- Typo here?
>				(S_AXI_ARLEN    ? S_AXI_ARLEN  : rlen), <= and here?
>				next_rd_addr);

For i_size and i_last_addr, this decides whether to use the current or latched version based on S_AXI_ARREADY, which makes sense. But for i_burst and i_len it decides current or latched based on the current value on the bus, which is incorrect. It means we are looking (and caring about) the values of the AR* signals when ARVALID && ARREADY is not true, this violating one of the fundamental rules of AXI. 

As currently coded, if I request a 12-byte, INCR burst, and then reset all of the AR* data lines to X as soon as the transaction is accepted, then instead of an incrementing burst, I get the address (o_raddr) incrementing once, but all subsequent reads in the burst just re-use the same address. See simulation trace below. 

[image]

ZipCPU, 2019-06-10:
Good catch!  Thank you.  Look for a patch tomorrow.

ZipCPU, 2019-06-11:
Fixed in [42b0c84](https://github.com/ZipCPU/wb2axip/commit/42b0c84d0b5cca191acd99a62b61ab983f163012)

closed 2019-06-11 by ZipCPU