ZipCPU/wb2axip #11: axi2axilite.v: M_AXI_BVALID does not propagate to S_AXI_BVALID

bchetwynd, 2019-06-11:
Performing a AXI4 to WB conversion using the recommended components (per the repo's README.md)

Issued a single write, which successfully propagated to the WB target.  The axlite2wbsp component successfully generates the the assertion of M_AXI_BVALID, but I never see S_AXI_BVALID asserted.

Waveforms attached.  Parameters set to default with the exception of LGFIFO = 1

[image]
[image]

bchetwynd, 2019-06-11:
Initial user error: S_AXI_AWLEN and S_AXI_AWSIZE were set incorrectly (255 and 7).  Still debugging.

ZipCPU, 2019-06-11:
I just got to that point.  ;)

Keep going!  I've enjoyed your feedback so far.

bchetwynd, 2019-06-11:
Thank you.

It seems as if all the conditions to generate s_axi_bvalid seem be one cycle off.  I observe that the conditions controlling the increment and decrement of bcounts are occurring on the same cycle.  

Is my WB peripheral responding too quickly?

[image]
[image]

bchetwynd, 2019-06-11:
I should note that all my transfers are one word in size..... and thus the first and only data sent over S_AXI_WDATA has S_AXI_WLAST asserted.

bchetwynd, 2019-06-11:
The bcounts logic (axi2axilite.v, lines 320-328) does not handle simultaneous assertions of read_from_wrfifo and (skidm_bvalid and skidm_bready).  As a result, bcounts will just continue to increment and never decrement.

bchetwynd, 2019-06-11:
Also suggest updating axi2axilite.v, line 344 to the following:
				||((bcounts == 0) && (!wfifo_empty) && (wfifo_bcount == 1));
from:
				||((bcounts == 0) && (!wfifo_empty) && (wfifo_bcount == 0));

ZipCPU, 2019-06-11:
I'm not sure I follow your comment about lines 320-328 not  handling simultaneous read_from_wrfifo and (skidm_bvalid & skidm_bready).  read_from_wrfifo = some_expression & (skidm_bvalid & skidm_bready), so it must always handle skidm_bvalid && skidm_bready.  See for example [line 308](https://github.com/ZipCPU/wb2axip/blob/8ae43172c2f3640105e9ce5fc9beb39c3f17837c/rtl/axi2axilite.v#L308).  As for `skidm_bready`, that will only ever be true of the outgoing `S_AXI_B*` channel isn't stalled, see for example [line 375](https://github.com/ZipCPU/wb2axip/blob/8ae43172c2f3640105e9ce5fc9beb39c3f17837c/rtl/axi2axilite.v#L375).

Dan

ZipCPU, 2019-06-11:
As for your off by one error and your comment about bcounts checking for one rather than zero, the length field used by AXI is off by one.  Therefore, when you set AWVALID and AWLEN==1, you are asking for a burst of length two--not a burst of length one.  It is then an error to immediately set WVALID and WLAST.  That will force the return to wait an additional cycle before setting BVALID.

You'll want to set AWLEN to zero if you only want single value bursts.

Dan

bchetwynd, 2019-06-12:
Thank you.  Things behave as expected when I pass correct parameters aw_len = 0; aw_size = 4.

closed 2019-06-12 by bchetwynd