ZipCPU/wb2axip #13: `default_nettype none causes issues when integrating with Xilinx IP

bchetwynd, 2019-06-19:
My build process is scripted and I cannot modify the Xilinx IP that attempts to integrate with these components.

I therefore have temporarily removed the `default_nettype none deceleration from these files and synthesis is able to progress (Vivado 2016.4), but the issue also shows up with Vivado 2018.3

- Brendon

ZipCPU, 2019-06-19:
The issue is that according to the verilog standard, the synthesizer is supposed to treat all source files as if they were concatenated together.  This macro then forces other files to be treated with the same standard.  Within your design, that should be a good thing.  When using someone else's IP, this can be a problem.

One solution is to place the line `default_nettype wire` at the end of the file.  That sets the synthesis tool back to its original behavior of ignoring variable names that haven't been first defined, and treating anything undefined as a "wire".

bchetwynd, 2019-06-19:
Dan, that seems like a reasonable fix.  Apparently, for some IP, xilinx does not define their net types.

ZipCPU, 2019-06-19:
I had this problem originally with Quartus, so it's not limited to Xilinx.

Feel free to pencil it in, and I'll hold this issue open until I add the changes in properly.

ZipCPU, 2019-06-19:
Try #14 and see if it does what you need,
Dan

bchetwynd, 2019-06-20:
Dan,

That does exactly what I need.  Thank you.

closed 2019-06-20 by bchetwynd