ZipCPU/wb2axip #15: xlnxdemo formal verification fails

bluewww, 2019-08-14:
When I run `sby -f xlnxdemo.sby` with the FOSS yosys version (just built the master branch at 19d6b8846f55b4c7be705619f753bec86deadac8)

I get this:
```
SBY 14:52:58 [xlnxdemo_cvr] Removing direcory 'xlnxdemo_cvr'.
SBY 14:52:58 [xlnxdemo_cvr] Copy 'faxil_slave.v' to 'xlnxdemo_cvr/src/faxil_slave.v'.
SBY 14:52:58 [xlnxdemo_cvr] Copy 'xlnxdemo.v' to 'xlnxdemo_cvr/src/xlnxdemo.v'.
SBY 14:52:58 [xlnxdemo_cvr] engine_0: smtbmc
SBY 14:52:58 [xlnxdemo_cvr] base: starting process "cd xlnxdemo_cvr/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 14:52:59 [xlnxdemo_cvr] base: finished (returncode=0)
SBY 14:52:59 [xlnxdemo_cvr] smt2: starting process "cd xlnxdemo_cvr/model; yosys -ql design_smt2.log design_smt2.ys"
SBY 14:52:59 [xlnxdemo_cvr] smt2: finished (returncode=0)
SBY 14:52:59 [xlnxdemo_cvr] engine_0: starting process "cd xlnxdemo_cvr; yosys-smtbmc --presat --unroll -c --noprogress -t 60 --append 0 --dump-vcd engine_0/trace%.vcd --dump-vlogtb engine_0/trace%_tb.v --dump-smtc engine_0/trace%.smtc model/design_smt2.smt2"
SBY 14:52:59 [xlnxdemo_cvr] engine_0: ##   0:00:00  Solver: yices
SBY 14:52:59 [xlnxdemo_cvr] engine_0: ##   0:00:00  Checking cover reachability in step 0..
SBY 14:52:59 [xlnxdemo_cvr] engine_0: ##   0:00:00  Checking cover reachability in step 1..
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Reached cover statement at xlnxdemo.v:798 in step 1.
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to VCD file: engine_0/trace0.vcd
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to Verilog testbench: engine_0/trace0_tb.v
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to constraints file: engine_0/trace0.smtc
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Checking cover reachability in step 1..
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Reached cover statement at xlnxdemo.v:788 in step 1.
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to VCD file: engine_0/trace1.vcd
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to Verilog testbench: engine_0/trace1_tb.v
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to constraints file: engine_0/trace1.smtc
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Checking cover reachability in step 1..
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Checking cover reachability in step 2..
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Reached cover statement at xlnxdemo.v:1019 in step 2.
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to VCD file: engine_0/trace2.vcd
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to Verilog testbench: engine_0/trace2_tb.v
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to constraints file: engine_0/trace2.smtc
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Checking cover reachability in step 2..
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Reached cover statement at xlnxdemo.v:901 in step 2.
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to VCD file: engine_0/trace3.vcd
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to Verilog testbench: engine_0/trace3_tb.v
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Writing trace to constraints file: engine_0/trace3.smtc
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Checking cover reachability in step 2..
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:00  Checking cover reachability in step 3..
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Reached cover statement at xlnxdemo.v:902 in step 3.
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to VCD file: engine_0/trace4.vcd
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to Verilog testbench: engine_0/trace4_tb.v
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to constraints file: engine_0/trace4.smtc
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Checking cover reachability in step 3..
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Reached cover statement at xlnxdemo.v:1020 in step 3.
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to VCD file: engine_0/trace5.vcd
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to Verilog testbench: engine_0/trace5_tb.v
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to constraints file: engine_0/trace5.smtc
SBY 14:53:00 [xlnxdemo_cvr] engine_0: ##   0:00:01  Checking cover reachability in step 3..
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Checking cover reachability in step 4..
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Reached cover statement at xlnxdemo.v:1021 in step 4.
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to VCD file: engine_0/trace6.vcd
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to Verilog testbench: engine_0/trace6_tb.v
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to constraints file: engine_0/trace6.smtc
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Checking cover reachability in step 4..
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Reached cover statement at xlnxdemo.v:903 in step 4.
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to VCD file: engine_0/trace7.vcd
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to Verilog testbench: engine_0/trace7_tb.v
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to constraints file: engine_0/trace7.smtc
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Checking cover reachability in step 4..
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Checking cover reachability in step 5..
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Reached cover statement at faxil_slave.v:704 in step 5.
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to VCD file: engine_0/trace8.vcd
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to Verilog testbench: engine_0/trace8_tb.v
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Writing trace to constraints file: engine_0/trace8.smtc
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:01  Checking cover reachability in step 5..
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:02  Reached cover statement at xlnxdemo.v:904 in step 5.
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:02  Writing trace to VCD file: engine_0/trace9.vcd
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:02  Writing trace to Verilog testbench: engine_0/trace9_tb.v
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:02  Writing trace to constraints file: engine_0/trace9.smtc
SBY 14:53:01 [xlnxdemo_cvr] engine_0: ##   0:00:02  Checking cover reachability in step 5..
[... 260 lines cut ...]
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Checking assertions in step 4..
SBY 14:53:29 [xlnxdemo_prf] engine_0.induction: ##   0:00:00  Trying induction in step 34..
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Checking assumptions in step 5..
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Checking assertions in step 5..
SBY 14:53:29 [xlnxdemo_prf] engine_0.induction: ##   0:00:00  Trying induction in step 33..
SBY 14:53:29 [xlnxdemo_prf] engine_0.induction: ##   0:00:00  Trying induction in step 32..
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Checking assumptions in step 6..
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Checking assertions in step 6..
SBY 14:53:29 [xlnxdemo_prf] engine_0.induction: ##   0:00:00  Trying induction in step 31..
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  BMC failed!
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Assert failed in xlnxdemo: xlnxdemo.v:774
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Writing trace to VCD file: engine_0/trace.vcd
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Writing trace to Verilog testbench: engine_0/trace_tb.v
SBY 14:53:29 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Writing trace to constraints file: engine_0/trace.smtc
SBY 14:53:30 [xlnxdemo_prf] engine_0.basecase: ##   0:00:00  Status: FAILED (!)
SBY 14:53:30 [xlnxdemo_prf] engine_0.basecase: finished (returncode=1)
SBY 14:53:30 [xlnxdemo_prf] engine_0: Status returned by engine for basecase: FAIL
SBY 14:53:30 [xlnxdemo_prf] engine_0.induction: terminating process
[... cut ...] ummary: Elapsed clock time [H:MM:SS (secs)]: 0:00:02 (2)
SBY 14:53:30 [xlnxdemo_prf] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:02 (2)
SBY 14:53:30 [xlnxdemo_prf] summary: engine_0 (smtbmc) returned FAIL for basecase
SBY 14:53:30 [xlnxdemo_prf] summary: counterexample trace: xlnxdemo_prf/engine_0/trace.vcd
SBY 14:53:30 [xlnxdemo_prf] DONE (FAIL, rc=2)
```
Unfortunately I'm too much of a beginner to figure out where the issues is. I can also provide the vcd files if you can't reproduce.

ZipCPU, 2019-08-14:
The issue is that [Xilinx's demo code is broken in the first place](http://zipcpu.com/formal/2018/12/28/axilite.html).  This is why I wrote the [demoaxi.v](http://zipcpu.com/blog/2019/01/12/demoaxilite.html) core--to fix the latent problems within their core, and to achieve higher throughput.

I can add some information to the [README](/ZipCPU/wb2axisp/blob/master/README.md) to point that out.

Dan

bluewww, 2019-08-14:
Ah thanks for the quick answer, so I missed that part in the blog post :)

closed 2019-08-14 by bluewww

ZipCPU, 2019-08-14:
See the updated note in the readme.

Dan