ZipCPU/wb2axip #19: Regarding err_state in WB to AXI 4 Lite Bridge

Nishikant-github, 2020-03-10:
Hi,
When I was running write-read-write transaction then for 3rd i.e. write transaction I am not getting output (means ack or err any one is not generated) . can you help me in that . I have attached the screenshot. please check and let me know.
If someone provides me specification it will be very helpful.
[image]
[image]

ZipCPU, 2020-03-10:
Do you have the core connected properly?  I would note that AXI_ARVALID is high, and AXI_ARREADY is low in your second figure, while the WB_STALL is low.  Looking over the internal logic, WB_STALL should be high if this ever takes place.

I would also note that the RVALID signal coming from your core is undefined for way to long.  That should be a clear zero following any reset.

Dan

Nishant-gith, 2020-03-11:
Hi Dan ,
Actually In my core I have not used WB_STALL signal . And I want to ask AXI_ARVALID  and AXI_ARREADY both should be high for complete transfer or AXI_ARREADY will be high for one clock cycle. can you clarify my doubt.

ZipCPU, 2020-03-11:
>Actually in my core I have not used WB_STALL signal.

This would be your problem then.

This core uses WB B4 (pipelined), not B3 (classic).  You can convert from the one to the other using a bridge called wbc2pipeline if you want, but you can't ignore STALL.  B4 (pipelined) is the faster protocol, capable of carrying out 100% throughput across multiple transactions.  B3 (classic) can only ever handle one outstanding transaction at a time.

In WB B4, a request is made every time STB && !STALL.  If you hold STB high while ignoring STALL, you are likely going to make multiple bus requests.  When you then drop CYC, outstanding requests are dropped and their responses are then ignored.  I could tell you more of what was going on in the traces above if STALL were present.  Without a valid STALL signal it looks like you are making 6 requests every time STB and CYC go high.  You wait for the first response, and drop CYC.  The other 5 requests are then left in an indeterminate state.

In AXI, anytime a request is made, it *cannot be dropped*.  There's no equivalent to dropping CYC and resetting the channel.  Therefore, if you ever drop the CYC line, the core then drops any outstanding responses to requests that were in progress and ignores whatever responses might come back to them.  Once the bus has been flushed, it then sends whatever new value you might wish to send.  

In other words, the core is working as designed.

Dan

closed 2020-03-11 by ZipCPU

Nishant-gith, 2020-03-24:
Hi Dan,
Now I understand , Thanks a lot for clarifying my doubts !!!