ZipCPU/wb2axip #23: missing src/spec.tex

tcmichals, 2020-06-05:
missing src/spec.tex

_Originally posted by @tcmichals in https://github.com/ZipCPU/wb2axip/issues/22#issuecomment-639662493_

ZipCPU, 2020-06-05:
The file is not missing, it just doesn't exist.  There is no doc/src/spec.tex file (yet).

I've tossed around whether or not I want to write one, but instead I've tried to place more extensive file headers containing usage information.  Building a spec.pdf file at this point would just duplicate what's already in either the source files or the various README's.  Those should each have extensive (and sufficient) information in them.

Please let me know if you are looking for something particular and not finding it.

Dan

tcmichals, 2020-06-05:
OK. I'll close this issue.  My goal is understand some of the basics of the libraries, wishbone to AXI and create a python generator similar to . wb_intercon.  Also what a great library,  very good work.

ZipCPU, 2020-06-05:
I've used [AutoFPGA}(https://github.com/ZipCPU/autofpga) as a bus generator with these files.  They're really designed for that purpose, although they should work with any other core generator.

Be careful, when using wb_intercon, the Wishbone files here use Wishbone pipeline, *not Classic*.  There is a bridge here to go from pipeline to classic--although your performance will go down should you use it.

closed 2020-06-05 by ZipCPU