ZipCPU/wb2axip #25: aximwr2wbsp increments address too early, first data written to the next incremented address

motylewski, 2020-08-10:
Simulated using Vivado 2019.2 AXI Verification IP 1.1
Data is always landing in incremented address. Both in HW and in the simulation.

In the test mit HW single  write there is the same problem - writing to addr+1 instead of addr.

[image]

ZipCPU, 2020-08-10:
I think I see it.

Check [this patch](https://github.com/ZipCPU/wb2axip/commit/2b7af47e14c73277191f05611c205357d4a9a483), and see if it fixes your issue.

Dan

motylewski, 2020-08-10:
Simulation looks OK now, thanks !
In HW also OK now.
But there is (and always was) another issue: endianess is swapped when writing. Reading is OK.

[image]

ZipCPU, 2020-08-10:
Fixed in [864d554](https://github.com/ZipCPU/wb2axip/commit/864d554399419614a37286855a38fa538ca5411c).

Dan

motylewski, 2020-08-11:
OK, thanks.
As for OPT_SWAP_ENDIANNESS I have added that into axim2wbsp so it is adjustable in the same way for both WR and RD channels. I am using 0 there anyway.

closed 2020-08-11 by motylewski