ZipCPU/wb2axip #27: always block with constant assignment

eschmidscs, 2020-08-18:
Hi

Disclaimer: I'm rather VHDL... The last Verilog training is some years ago. So please consider this a question rather than some report.

I'm trying to get the axilxbar running for a testbench simulated with QuestaSim. It seems that processes like (axilxbar.v line 854)
```Verilog
always @(*)
  mrindex[M] = 0;
```
are never evaluated, mrindex stays all x.
If I add an initial statement setting mrindex to zero, it suddenly works.

As there is no signal read in the process, the simulator might well think that it never has to schedule it. But is that a general problem or is this a QuestaSim related issue?
Do I hit this because I only have one master?

Best regards,
emanuel

eschmidscs, 2020-08-24:
Hi Dan,
ok, thanks for the background. I currently don't have time to fiddle around with QuestaSim, but it would be interesting to dig a bit deeper. As mentioned, my Verilog background is thin.
I just replace the always with initial for now. It would also work to have both always and initial. But this is also a bit ugly.
Best regards!

ZipCPU, 2021-03-13:
@eschmidscs ,

It turns out this is a Verilog simulator issue, and the problem was that my simulator wasn't standards compliant.

I've now fixed as many of the `always @(*)` issues I could find in a quick check.  Feel free to let me know if you come across any others, and I'll fix them as well.

Dan

closed 2021-03-16 by ZipCPU