ZipCPU/wb2axip #31: illegal localparam in list of parameters

speq, 2021-03-03:
Hi,

It is illegal to use localparam in the parameter list.
For NC, it will report an error:
file: axixbar.v
                localparam      IW = C_AXI_ID_WIDTH,
                         |
ncvlog: *E,LOCALP (axixbar.v,103|11): illegal localparam in list of parameters [12.2(IEEE-2001)].
                localparam      AW = C_AXI_ADDR_WIDTH,
                         |
ncvlog: *E,LOCALP (axixbar.v,104|11): illegal localparam in list of parameters [12.2(IEEE-2001)].
                localparam      DW = C_AXI_DATA_WIDTH,
                         |

For VCS,
the localparma will have some misbehave. (not changed w/ parameter assign)

ZipCPU, 2021-03-03:
`localparam`s in the parameter list is a feature from the SV specification, introduced by at least 2012.  (I'm not sure if it was available in the 2005 specification).  It is most definitely legal, although not all tools support them.

To get the design to work with older tools, you can often just rename the `localparam` statements as parameters.  Just be aware, they are not to be adjusted externally.  You might also be able to adjust how your tool is handling the files--some tools just need a special option before accepting this syntax.

Dan

closed 2021-03-09 by ZipCPU

ZipCPU, 2021-04-24:
I just came across this problem with Quartus.  As a result, I've gone back and removed any unnecessary localparam's from the portlist and placed them into the main body of any modules using them.  This is a partial fix, however, so I'm still quietly adjusting any remaining issues by swapping localparam for parameter on a case by case and use by use basis.

Dan