ZipCPU/wb2axip #34: Which yosys version should I use?

jimmysitu, 2021-04-23:
Hi, All
I try make the formal verification in bench/formal with cmd
```bash
make all
```
I get stop with the following error when I use yosys e32cef40
```bash
sby -f skidbuffer.sby prfc
SBY 20:21:26 [skidbuffer_prfc] Removing direcory 'skidbuffer_prfc'.
SBY 20:21:26 [skidbuffer_prfc] Copy '../../rtl/skidbuffer.v' to 'skidbuffer_prfc/src/skidbuffer.v'.
SBY 20:21:26 [skidbuffer_prfc] engine_0: smtbmc
SBY 20:21:26 [skidbuffer_prfc] base: starting process "cd skidbuffer_prfc/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 20:21:26 [skidbuffer_prfc] base: skidbuffer.v:452: ERROR: Identifier `\default' is implicitly declared and `default_nettype is set to none.
SBY 20:21:26 [skidbuffer_prfc] base: finished (returncode=1)
SBY 20:21:26 [skidbuffer_prfc] base: job failed. ERROR.
SBY 20:21:26 [skidbuffer_prfc] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:21:26 [skidbuffer_prfc] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:21:26 [skidbuffer_prfc] DONE (ERROR, rc=16)
Makefile:721: recipe for target 'skidbuffer_prfc/PASS' failed
make: *** [skidbuffer_prfc/PASS] Error 16
```
But when I upgrade to yosys-0.9, I get
```bash
sby -f sfifo.sby prf
SBY 20:37:12 [sfifo_prf] Removing directory 'sfifo_prf'.
SBY 20:37:12 [sfifo_prf] Copy '../../rtl/sfifo.v' to 'sfifo_prf/src/sfifo.v'.
SBY 20:37:12 [sfifo_prf] engine_0: smtbmc
SBY 20:37:12 [sfifo_prf] base: starting process "cd sfifo_prf/src; yosys -ql ../model/design.log ../model/design.ys"
SBY 20:37:12 [sfifo_prf] base: finished (returncode=0)
SBY 20:37:12 [sfifo_prf] smt2: starting process "cd sfifo_prf/model; yosys -ql design_smt2.log design_smt2.ys"
SBY 20:37:12 [sfifo_prf] smt2: ERROR: No such command: dffunmap (type 'help' for a command overview)
SBY 20:37:12 [sfifo_prf] smt2: finished (returncode=1)
SBY 20:37:12 [sfifo_prf] smt2: job failed. ERROR.
SBY 20:37:12 [sfifo_prf] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:37:12 [sfifo_prf] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 20:37:12 [sfifo_prf] DONE (ERROR, rc=16)
SBY 20:37:12 One or more tasks produced a non-zero return code.
Makefile:682: recipe for target 'sfifo_prf/PASS' failed
make: *** [sfifo_prf/PASS] Error 16
```

ZipCPU, 2021-04-23:
I'm using: `Yosys 0.9+3855 (git sha1 54294957, clang 10.0.0-4ubuntu1 -fPIC -O2` and not having any problems.  That's a version with a date stamp of Jan 22, 2021.

Dan

closed 2021-04-24 by ZipCPU

jimmysitu, 2021-04-25:
Thanks, @ZipCPU, I can run the formal check with the yosys verision you provide