ZipCPU/wb2axip #38: axi2axilite

forflo, 2021-08-17:
Hi!

first and foremost, thank you for your wonderful work!

I am currently employing some of your designs in order to simulate Verilog code generated by Vivado HLS. For some C inputs, the HLS generates RTL that locks up the whole FPGA and I am trying to find out whether this is a case of me not having understood the HLS good enough (wrong settings for bus latency, etc.) or whether this is a bug in the generated AXI Master implementation.

My first setup worked without problems and the HLS IP produced valid results:

```
 .───────────────. 
(  Verilator TB   )
 `───────────────' 
         m         
         │         
      axilite      
         │         
         s         
         ▼         
    ┌────────┐     
    │ HLS IP │     
    └────────┘     
         m         
         │         
        axi        
         │         
         s         
         ▼         
 ┌──────────────┐  
 │ axi2axilite  │  
 └──────────────┘  
         m         
         │         
      axilite      
         │         
         s         
   ┌───────────┐   
   │ demoaxi.v │   
   └───────────┘   
```

The Verilator TB starts (and waits for) the HLS IP via axilite and the bundled AXI master ports of the HLS IP are connected to memory via axi. I increased the address width of `demoaxi.v` and `axi2axilite.v` to 32 and also increased the 64 memory units in `demoaxi.v` to 1MiB in order order to mimic a memory controller.

I then tried to introduce some latency between `axi2axilite` and `demoaxi.v` by way of a pipeline stage, because the system that the HLS IP works in employs an Xilinx AXI interconnect (with additional latency). Since I am trying to find out why the HLS IP locks up **that** system I have to mimic its latency. The pipeline stage is conceptually depicted below. (Note that the number of daisy-chained DFFs can be configured.)

```
                        ┌─────────┐                         ┌────────────────┐                    
                     ┌s─│ HLS IP  │─m                   s┌─▶│ pipeline stage │─m─┐                
                     │  └─────────┘ │                    │  └────────────────┘   │                
                     │              │                    │                       │                
                     │             axi                axilite                 axilite             
                 axilite            │                    │                       │                
                     │              │                    │                       │                
 .───────────────.   │              │  ┌──────────────┐  │                       │   ┌───────────┐
(  Verilator TB   )◀─m              s─▶│ axi2axilite  │m─┘                       s──▶│ demoaxi.v │
 `───────────────'                     └──────────────┘                              └───────────┘
------------------------------------------------------------------------------------
                                                                               ┌──────────────┐
┌──────────────────┐                                                           │              │
│                  │                                                           │              │
│                  │                                                           │              │
│                  │                 ┌──────────────────────┐                  │              │
│                  │ ───────────────▶│ clock         DataOut│──────AWADDR────▶ │              │
│                  │                 │                      │                  │              │
│                  │ ────AWADDR─────▶│ DataIn               │                  │              │
│                  │                 └──────────────────────┘                  │              │
│                  │                                                           │    demoaxi.v │
│axi2axilite.v     │                                                           │              │
│                  │                        ┌────────┐                         │              │
│                  │                        │  ...   │                         │              │
│                  │                        └────────┘                         │              │
│                  │                                                           │              │
│                  │                                                           │              │
│                  │                                                           │              │
│                  │                 ┌──────────────────────┐                  │              │
│                  │ ───────────────▶│ clock         DataOut│◀────ARREADY───── │              │
│                  │                 │                      │                  │              │
│                  │ ◀──ARREADY──────│ DataIn               │                  │              │
│                  │                 └──────────────────────┘                  │              │
│                  │                                                           └──────────────┘
└──────────────────┘                                                                           
```

With the additional latency, the HLS IP locked itself up. As far as I can tell from the simulation, the axi2axilite block does not return a `BVALID` to the HLS IP. I tried to increase `LGFIFO` to 4, 5, 6, and 32 which all yield the same results (a dead-lock). The `demoaxi.v` seems to return the write ack back to the axi2axilite though.

Is this a known limitation of the axi2axilite block? If so, how can I contribute?

I have a working symbiyosys setup and a little experience with it. Would it help if I try to add my axilite pipeline stage to increase the latency of the instance `faxil_master` in the `FORMAL` section of `axi2axilite.v`?

Best regards,
Florian

ZipCPU, 2021-08-17:
Without more information, it would be hard to comment on the problem.  Here are some things you can look for though:

1. I've been having some recent problems with 'x propagation and some Verilog simulators.  Please check your traces to see if any of the AXI signals are marked as 'x.  If so, I have some updates I can push to the `skidbuffer.v` which might help.  Indeed, this would be my first suspicion.  If this is the case, it's an easy fix--but it does require finding the root cause of the 'x propagation.  (Look for always @(*) X = CONSTANT; blocks--these work in hardware and Verilator, just not in many Verilog simulators.)
2. If possible, I would like to visually check your delay component.  Did you add it to axi2axilite.v, or is it external?  Is it a separate entity?  If so, and you could post a github gist or some such, I'll take a look at it.  If it's a simple delay, it should be easy to verify.

You should know that verifying the axi2axilite.v module was a real pain to accomplish.  Subtle changes to it are likely to be similarly difficult to verify, so my recommendation would be to make them external to the axi2axilite.v module.

There should be no reason to increase the FIFO size within the axi2axilite.v.  I mean, the only thing that might benefit from doing so would be a slave with a very deep pipeline.  These aren't very common, so the smaller the FIFO size the better--at least until it starts impacting throughput, which is why I have it set to what it is.

I should also point out that even today I'm using the axi2axilite.v and demofull.v components in my simulations.  These have no pending updates, and are working quite nicely for me.  The skidbuffer, on the other hand, may be the culprit ... so take a peek and see what you can see from the trace.  Are there 'x values?  Are they coming from the skidbuffer?  If so, I can make a quick push with the updates I have locally.

Dan

forflo, 2021-08-18:
Hi,

thank you for your very quick answer!

> I've been having some recent problems with 'x propagation and some

I've looked into it, but did not find anything yet. I do not see any 'x propagation, probably because I am only using Verilator.

> If possible, I would like to visually check your delay component.

I assembled a reproduction package [here](https://github.com/forflo/verilaxi). The delay component sits in [here](https://github.com/forflo/verilaxi/blob/master/axilite_pipeline_stage.v) and references [this](https://github.com/forflo/verilaxi/blob/master/pipeline_stage.v) module.

> be easy to verify.

 At the bottom of `axilite_pipeline_stage` I tried to [formally verify](https://github.com/forflo/verilaxi/blob/master/formal_driver.sby) that the delay stage works as intended. I did not prove the core yet.

I have attached the vcd file [dl_dump.vcd.zip](https://github.com/ZipCPU/wb2axip/files/7009410/dl_dump.vcd.zip) to this post. The HLS IP signals (where the HLS IP acts as master are `TOP.top.hls_ip.m_axi_DATA_*`. These signals are wired into `TOP.top.ram_inst` which is an instance of `axi2axilite.v`. For `latency = 1` in [the pipeline stage](https://github.com/forflo/verilaxi/blob/dd41a0c826a77ec7f42b537b2dc09bb5c8e38932/axilite_bram_pipelined.v#L3), the `axi2axilite.v` block does not return a BREADY back to the HLS IP.

In the attached screenshot you can see that the HLS IP starts a write burst at 84ns. In 96ns, the IP then does a burst read (while the write burst is still going). WLAST of the write burst happens at 132ns (when the read has already been finished. **The strange thing that I do not understand is that in 136ns, the S_AXI_WREADY of the unmodified `demoaxi.v` goes to 0.** The memory dump of the `demoaxi.v` tells me that only the first axilite write was successful.

Is it disallowed to interleave AXI write and AXI reads?

[image]

ZipCPU, 2021-08-19:
The problem lies in your pipeline stage.

You can see the bug in the trace, looking at the AXI interface of your bram_inst, where S_AXI_AWVALID && S_AXI_AWREADY on the first cycle, S_AXI_AWVALID && !S_AXI_AWREADY on the second cycle, and then !S_AXI_AWVALID on the third.  This is a protocol violation.  S_AXI_AWVALID *MUST* remain high until S_AXI_AWVALID && S_AXI_AWREADY.

The bottom line is that it is illegal to just delay the AXI xVALID signal independent of what the associated xREADY is doing.  Worse, you are delaying the xVALID, xREADY, and their associated data signals all independent of one another.  There are handshaking rules that are getting violated all over the place by doing this.

How might you fix this?

If you just want to keep an N-cycle delay, then a better latency stage could be generated by stringing together several skidbuffers--just make sure to set the OPT_OUTREG to 1 in order to get a guaranteed delay of one clock cycle per skid buffer.  Placing two skid buffers in a row, with valid, read, and data connected between them, will give you a latency of two cycles, etc.  Notice that the associated data signals must also be delayed with the VALID.  So, when delaying AWVALID, for example, you'll have to also delay AWADDR and AWPROT--something missing from your pipeline delay stage.

On the other hand, if you would rather randomly delay a single burst at a time, you can insert a `stall` signal in the handshake.  AND the upstream READY with the `!stall` signal as well as the downstream `VALID`.  You can then set the `stall` signal as you wish to create stalls in the protocol.  This isn't exactly the same as your pipeline delay, but you can use it to delay a VALID/READY handshake.  Do be aware that this is illegal to do in practice--due to the rule that AXI outputs are not to be driven combinatorially by AXI inputs, but it works well enough in simulation.  I think the rule is there to make sure timing isn't impacted one way or another, something which isn't usually an issue in simulations.  If you use this approach, you won't need to also delay the associated data signals.  The end result will be a throughput delay, not a latency delay.  You're the engineer--your call.

Oh, and the AXI2AXILITE module works as designed: It issues two write requests, and gets only one BVALID in return.  It then hangs waiting for the second BVALID according to the AXI4-lite protocol.  Therefore, I'm going to close this issue.

Dan

closed 2021-08-19 by ZipCPU