ZipCPU/wb2axip #41: AXIS2MM & WUSER bits/byte causes critical warning in Vivado

baileyji, 2022-01-19:
Hi Dan, 
First a disclaimer this may well not be an issue with axis2mm.v and merely a cosmetic interplay with Vivado's IP Validation system. I'm trying to test the core in a design I'm presently having HLS issues with the same task. The inbound stream consists of TLAST, TDATA (256b), and no TUSER  (TKEEP, and TSTRB are there but ripped out in implementation as I don't use them and HLS insists on placing them on blocks).

I've configured your core, whose output is routed to a SmartConnect with a null TUSER width but I see in the verilog that that still results in a single wire for TUSER in and out. This results in a critical warning in validation (which I'm attempting to ignore):
`[xilinx.com:ip:smartconnect:1.0-1] capture_play_axi_smc_1: WUSER_WIDTH (1) of S00_AXI must be integer number of bits per byte of WDATA (256).` And as I understand it that warning is based on the AXI4 standard. 
 
I suspect this isn't strictly a bug, but given the effort you've put into these blocks (but have not tested this one in a design per your docs) I thought you might be interested. 

I've got your core configured as below, I'll be interested to see how it holds up with a 512MHz clock and a 4GiB max transfer of a 256b stream. 

```
		parameter	C_AXI_ADDR_WIDTH = 36,
		parameter	C_AXI_DATA_WIDTH = 256,
		parameter	C_AXI_ID_WIDTH = 1,
		parameter	C_AXIS_TUSER_WIDTH = 0,
		parameter [0:0]	OPT_TLAST_SYNC = 1,
		parameter [0:0]	OPT_TREADY_WHILE_IDLE = 1,
		parameter [7:0]	ABORT_KEY = 8'h26,
		parameter	LGFIFO = 9,
		parameter	LGLEN  = 32,
		parameter [C_AXI_ID_WIDTH-1:0]	AXI_ID = 0,
		parameter	[0:0]	OPT_LOWPOWER  = 1'b0,
		parameter	[0:0]	OPT_CLKGATE   = OPT_LOWPOWER
```

ZipCPU, 2022-01-19:
[Is this you](https://support.xilinx.com/s/question/0D52E00006wSqFTSA0/improving-20212-performance-for-axi-s2mm?language=en_US)?  Either way, welcome to the repo.

As for your request, the problem is somewhat of a tool/language issue.  First, I don't know a way to declare a zero width I/O.  Declaring something of width [0:0] creates a one bit wire, and a declaration of [-1:0] is either an error or it creates a two-bit connection.  This means that if the xUSER values are to exist at all they must have at least one bit of width--even if they aren't used.  I would normally declare these values as [C_AXIS_TUSER_WIDTH-1:0], however if the TUSER width is zero then this turns into a declaration of [-1:0].  This generates a synthesis error, or at least nasty lint warnings.  To avoid this problem, I declare the xUSER bits with a [0:0] width even when TUSER is zero.  This leads to the problem you have noted above.  That leaves us with a couple of options:

1. You could wrap this with another module, and thus remove the TUSER I/O's in the wrapper
2. You could modify the design to rip out the (unused) TUSER I/O's.  A simple internal declaration of the TUSER values, and setting the input TUSER to zero, should be sufficient.

    wire [0:0] S_AXIS_TUSER = 0;
    wire [0:0] M_AXI_WUSER;

Dan

baileyji, 2022-01-19:
It is. Thanks for 2, that's exactly what I needed. My grasp of verilog is very limited so when I saw the wire was used in an internal fifo I wasn't sure how to just rip it out. that cleared the warning.

closed 2022-01-19 by baileyji

ZipCPU, 2022-01-19:
So glad I could help.  I get the feeling, though, that your task is far from complete.  Let me know how it goes.  I'd love to know if the S2MM IP works within your task, or if you have to reject it for some reason and if so what that reason was.

Thanks!

Dan