ZipCPU/wb2axip #43: Vivado Block design, local param, and axixbar

baileyji, 2022-01-24:
In making progress with the axis2mm core I've run into a 100% duty cycle issue with both the xilinx SmartConnect and Interconnect cores, so while I'm waiting on forum posts re:that I thought I'd give your axixbar core a whirl and try building up my own. I think I'm running into something related to #18. I'm also seeing similar complaint regarding axivfifo. This is in Vivado 2021.1

axixbar:
```
 [IP_Flow 19-350] User Parameter 'SLAVE_ADDR (Slave Addr)': Missing default value
 [IP_Flow 19-350] User Parameter 'SLAVE_MASK (Slave Mask)': Missing default value
 [IP_Flow 19-350] HDL Parameter 'SLAVE_ADDR (Slave Addr)': Missing default value
 [IP_Flow 19-627] HDL Parameter 'SLAVE_ADDR (Slave Addr)': XPath expression failed: Undefined parameter "AW" used in expression "{"111",{(AW - 3){"0"}},"110",{(AW - 3){"0"}},"101",{(AW - 3){"0"}},"100",{(AW - 3){"0"}},"011",{(AW - 3){"0"}},"010",{(AW - 3){"0"}},"0001",{(AW - 4){"0"}},"0000",{(AW - 4){"0"}}}".
 [IP_Flow 19-350] HDL Parameter 'SLAVE_MASK (Slave Mask)': Missing default value
 [IP_Flow 19-627] HDL Parameter 'SLAVE_MASK (Slave Mask)': XPath expression failed: Unbalanced parentheses found in expression "((spirit:decode(id('MODELPARAM_VALUE.NS')) <= 1) ? "1111",{(AW - 4){"0"}}} : {{(spirit:decode(id('MODELPARAM_VALUE.NS')) - 2){"111",{(AW - 3){"0"}}}},{2{"1111",{(AW - 4){"0"}}}})".
```

I was able to work around these issues by hard coding what I thought would be correct for a 2 to 1 512b data, 39b addr xbar and place the block. However the result is a block that would be _very_ tedious to use as the axi interfaces are not separated properly. Compare the block with the xilinx crossbar:
[image]
[image]

ZipCPU, 2022-01-24:
Hmm, yes, I see your problem.

Let's start at the top, though, with the localparam's.  I'll take a note to look into that one.  It'd be annoying, 'cause it'll make the line lengths all that much longer, but let me take a peek at what can be done there.

As for how tedious the interface can be to use, yeah, I can get that one as well.  That's why I connect it up using software, rather than by hand using Xilinx's IP integrator.  At this, I'm open to ideas.  However, Xilinx's approach is a non-starter--it would break all kinds of lint rules.  For example, I/O's must be used, they cannot be left unconnected.  Likewise, I/O's must use the full width.  etc., etc., etc.  Perhaps there's an SV way to do this?  Perhaps I should write a script for this purpose?  But ... [I have one, and it is open source](https://github.com/ZipCPU/autofpga) ... so I don't really have a good answer to this.

Dan

baileyji, 2022-02-01:
Thanks for the feedback. I'll close this as ideas here are beyond what I can offer.

closed 2022-02-01 by baileyji