ZipCPU/wb2axip #54: is the axiclk can be used in the axi lite scenario

yuanjin0928, 2023-10-10:
Hi,

I am wondering if I can connect the interface of the axiclk directly to a axi lite interface.

Best,
Jin

ZipCPU, 2023-10-12:
I'm not sure I follow your question.  Can you explain more?  Are you asking about the axixclk IP, used for crossing clock domains or something else?

Dan

yuanjin0928, 2023-10-17:
> I'm not sure I follow your question. Can you explain more? Are you asking about the axixclk IP, used for crossing clock domains or something else?
> 
> Dan

yuanjin0928, 2023-10-17:
Yes. I want to know if axiclk ip can be connected to with two axi lite interfaces which are in different clock domain.

ZipCPU, 2023-10-19:
If you only connected the AXI-Lite signals and either grounded or ignored all the other signals, then yes, it would work.  It would also generate a mess of lint warnings.  You might find yourself happier if you removed all of the references to the signals that are not AXI-Lite and rebuilt.

Frankly, the axixclk IP is so simple I often wonder why I have it.  All it does is run the separate AXI channels through asynchronous FIFOs.  You could do the same without using the axixclk IP.

Dan

ZipCPU, 2023-10-24:
This issue appears to be closed.

closed 2023-10-24 by ZipCPU