ZipCPU/wb2axip #56: Arbitration Behaviour of axixbar

predator2k, 2023-11-13:
In axixbar.v it says

//	Note that this arbiter only forwards AxID fields.  It does not use
//	them in arbitration.  As a result, only one master may ever make
//	requests of any given slave at a time.  All responses from a slave
//	will be returned to that known master.  This is a known limitation in
//	this implementation which will be fixed (in time) with funding and
//	interest.  Until that time, in order for a second master to access
//	a given slave, the first master must receive all of its acknowledgments.

I wonder if there's a "protection mechanism" or not, e.g. 

1. when one AXI master is waiting for the read response(of its read request) the other AXI master's ARREADY and AWREADY signals are de-asserted.
2. ARREADY/AWREADY are de-asserted unless ARVALID/AWVALID is asserted (otherwise two AXI masters can issue read requests and xbar will accept them together).

Or it's the AXI masters' responsibility to not issue requests at the same time?

Looking forward to your reply.

ZipCPU, 2023-11-21:
Let's see what I can answer here:

> When one AXI master is waiting for the read response(of its read request) [will the other] AXI master's ARREADY and AWREADY signals [get] de-asserted?

Not quite.  The read channel doesn't impact write arbitration.  Hence, while one master is interacting with the read channel, another master might be interacting with the write channel.  Second, each channel has a skidbuffer on entry.  As a result, all channels hold AxREADY high until they have accepted a single request.  Once that request has been entered, AxREADY will typically drop until that channel receives arbitration.

> [Will] ARREADY/AWREADY [be] de-asserted [until] ARVALID/AWVALID [are] asserted?

Not at all.  Because of the skid buffers, the AxREADY signals are held high.  If the crossbar isn't ready to accept a beat on the same cycle, it will get buffered until it can be internally forwarded.

>  Or it's the AXI masters' responsibility to not issue requests at the same time?

The AXI master has only one responsibility: follow the AXI specification.  No other requirements are made upon the various masters connected to the crossbar.

Dan

predator2k, 2023-11-22:
I see. So the core idea is that all masters have their own skid buffer to save the request, and AxREADY mostly depends on whether the skid buffer is full or not. Am I right?

ZipCPU, 2023-11-22:
Exactly.

predator2k, 2023-11-22:
OK. Thank you for your reply & good work.

closed 2023-11-22 by predator2k