ZipCPU/wb2axip #59: fusesoc file based on CAPI 1 standard.

johnathan-convertino-afrl, 2023-12-14:
fusesoc has moved to the CAPI 2 standard for its core description files. The current core file wb2axip.core is CAPI 1 and will cause fusesoc to give an error to the end user. Recommend removing it to avoid confusion.

Future Improvement: individual fusesoc core files in the repo, if this is something you would like let me know and I'll get it done.

ZipCPU, 2023-12-18:
Sigh.

This is one of my frustrations with FuseSOC.  While I love the project's goals, and think very highly of its developer, I don't use it.  This makes it difficult for me to maintain the core files used by it, as I never know when something changes, or what it needs to change to in order to keep a functionality I'm not using working.  Likewise, I have no means of automatic testing to verify that the FuseSOC core's continue to work as desired.

johnathan-convertino-afrl, 2023-12-19:
I haven't personally had that issue in version 2 (started at 2.1 and now up to 2.3).

Thought the documentation is frustrating to say the least. I end up reading source code to figure out how to use it. This is really not acceptable.

I have added fusesoc core files for some this repos IP in a separate repo. Essentially it uses a generator to git pull your repo so that the files are available to the core files in mine. Only issue is fusesoc finds your old core file and complains. Which is why I put in the request to delete it, since it seemed to serve no purpose at this point.

[repo link](https://github.com/johnathan-convertino-afrl/zipcpu_wb2axip_fusesoc)

This method keeps the files from tainting this repo.