ZipCPU/wb2axip #63: AXIS2MM Flushing FIFO

chrissbarr, 2024-12-20:
Hi,

This is not an issue as such, more a question on the behaviour of the AXIS2MM component. Please feel free to close this if this is out of scope here.

I've recently swapped to using this AXIS2MM implementation to replace the Xilinx DMA IP block, and am finding this implementation to be hugely more reliable, predictable, and simpler to configure. All round it has been a huge upgrade - I had no end of problems with the Xilinx DMA.

As I am currently using it, I am streaming 20-word long 'packets' in on the AXIS. I have TLAST that is asserted with the last word in the packet.

I have the AXIS2MM component configured with:
- LGFIFO = 9 (512 word FIFO)
- LGLEN = 31 (2 GB maximum transfer size)
- OPT_TLAST_SYNC  = 1

My stream of packets may stop at any time, although I should always receive a whole packet (and so stream should always end with a TLAST).

I am configuring the AXIS2MM via the AXIL interface. I am configuring the CMD_ADDR and CMD_LEN to match a region of memory I have dedicated as a buffer. The amount of data I receive via the AXIS is variable, but always smaller than this buffer, so the AXIS2MM never completes the transfer.

The controlling firmware knows when the packet stream stops via a control signal. When the stream ends, I wait a few milliseconds, command the AXIS2MM to abort, and wait for the abort in progress flag to be de-asserted. I then read CMD_LEN to see the number of bytes remaining, and work out the number of bytes transferred from that.

I see the AXIS2MM write to the AXI bus whenever there are enough words in the FIFO to do a burst of 256 words. Because my packets are only 20 words long, there are often many packets worth of data still in the AXIS2MM's FIFO when the stream ends. My goal is to ensure that all packets are written to memory.

Is there any way I can command the AXIS2MM to write out the data remaining in the FIFO, even if there is not enough data to do a 256-word long AXI burst? Abort only concludes any currently active AXI transactions, it won't issue any new ones to flush the FIFO, is that correct?

Or is there a way I can configure an AXI burst to happen whenever a packet ends? I had originally interpreted the OPT_TLAST_SYNC parameter would achieve something like this, but I think I have misunderstood it as I don't observe a change in behaviour with either configuration.

My current solution is simply to reduce LGFIFO = 5 (32 words), so that the FIFO is only large enough to hold ~ 1.5 packets, and so the amount of data that I 'lose' is small. 

Is there a configuration of this component that should do what I'm trying here? Happy to be told that I'm using it wrong, and really appreciate any insight.

ZipCPU, 2024-12-20:
I saw this last night.  It's certainly not how I was expecting the S2MM IP to be used when I built it.  So ... I slept on this, to see if I could come up with a good answer for you.  Perhaps not, but here's what I've got.

The S2MM was built with the anticipation of knowing, ahead of time, how much data you would be wanting to transfer.  This is very different from knowing ahead of time the _maximum_ amount of data you wish to transfer, only to have that maximum limited by a TLAST (or other indication) at a later time.  Indeed, support for TLAST is (sadly) almost an afterthought.  That's why the TLAST synchronization option is there--it allows you to guarantee you are sync'd when you start--not when you complete.

I have [another AXI S2MM IP](https://github.com/ZipCPU/zipcpu/blob/7892c9385df30e45011b8ae759b1ef2d196e4e62/rtl/zipdma/zaxdma_s2mm.v) that I've been building for when TLAST is more critical to the application.  This seems to be closer to what you are trying to do.  In this case, TLAST terminates the transfer entirely, so you would therefore need some external work to make certain you removed TLAST from all packets until the final one.  It's a different design, in that it is intended to be scripted from a nearby state machine, rather than from an external AXI-Lite bus.  That state machine could be controlled via AXI-Lite.  The state machine's purpose is to control where the data is coming from, and hence the size of the transfer is determined elsewhere--allowing the transfer to terminate properly on TLAST.  However, it does require that you know which packet is the final one while it is received in order to set TLAST on it.  Sadly, [this other development](https://github.com/ZipCPU/zipcpu/blob/7892c9385df30e45011b8ae759b1ef2d196e4e62/rtl/zipdma/zaxdma_s2mm.v) is ongoing and, although it passes both formal and simulation checks, I'm working through customer complaints of the design failing in hardware--things that haven't yet been resolved.  (Could be the IP's fault, could be a fault in the customer's setup, don't know yet ...)

[Other packet work I've done](https://github.com/ZipCPU/eth10g/blob/master/rtl/net/pkt2mem.v) attempts to maintain packet boundaries, by writing them to memory.  That IP doesn't (yet) have an AXI version (it's wishbone only for now).  Still, it doesn't sound appropriate for this application.

As for the abort, the purpose of the abort was to allow a user to "reset" the IP without resetting the entire bus and everything on it.  As a result, the best it does is a rather messy shutdown--while guaranteeing that the bus will be safe.  (Resetting the S2MM by itself, while in operation, would likely crash the entire AXI bus--something others have struggled with when using other S2MM designs.)  That's why there's no guarantee that the contents of the FIFO get written before completing the abort operation.

So ... 1) I see your problem, 2) that's not how this IP was intended to be used, 3) I don't think I have anything that does exactly what you want out of the box, and 4) I'm not sure at this point what the best solution is for solving your problem.  Any thoughts or suggestions?

I'll leave the thread open for a while in case you'd like further discussion, but otherwise I will (sadly) consider this "issue" closed.

chrissbarr, 2024-12-21:
Hi Dan,

Firstly, thank you for responding so quickly and in such a thoughtful way!

It's useful to have confirmation that the way I am using this S2MM IP is not exactly as intended, and that the behaviour I was hoping for isn't currently in scope. The current behaviour is very reasonable based on the design intent of this IP as you've described. You've saved me a lot of trouble trying to work out if I am missing something in my configuration or usage, cheers.

Thanks also for the link to the other S2MM IP you're working on, that's very interesting and will be a handy reference. I'll spend some time getting to grips with that.

At the end of the day, my application can live with losing some packets from the end of the stream, I just want to minimise that loss as much as possible. I have found this S2MM IP to be much more reliable than the previous Xilinx DMA IP, and I had the same shortcoming with that IP, so my current plan is to stick with this IP, albeit probably with the FIFO length reduced so I only lose a handful of packets.

My application is similar to a scope / logic analyzer. Reading your blog (great resource!) I find similarity to the WBSCOPE you give as an example when discussing bursting AXI masters [here](https://zipcpu.com/blog/2020/06/16/axiaddr-limits.html). Particularly the notes about supporting a partial burst when data logging ends sounds very similar to my problem. I think reading more into your implementation there might be a good avenue for me as well.

I have experimented with adding a 'flush' command that triggers the S2MM IP to begin a new AXI transaction even if the data in the FIFO is less than the configured AXI burst length. This appears to be working for me in simulation, I have not tested in hardware yet. The biggest problem I can identify with this so far is that it breaks the guarantee that future bursts will be aligned to the 4kB boundaries. So, if I was to trigger a flush, and then continue to stream new data into the S2MM, eventually an invalid boundary-crossing AXI burst would be generated.

In my application I will be doing a flush and abort in short sequence, so I might be able to get away with this limitation. I'd be interested in finding a safer / more general solution though. 

The changes I've made to do this are in the following diff file: [axis2mm_flush.diff.txt](https://github.com/user-attachments/files/18216849/axis2mm_flush.diff.txt)

Your page [about bursting AXI masters](https://zipcpu.com/blog/2020/06/16/axiaddr-limits.html) has some great notes on calculating burst length to avoid crossing 4kB boundaries. I'm going to spend some time seeing if I can get to grips with how the burst length calculations in this IP currently work, and if there's any scope to adjust them so that the boundaries are respected even if some of the burst lengths are variable.

If you have any further thoughts or pointers, I'd love to hear them. Otherwise, yes, please feel free to close this out whenever you like.

Thanks again!

ZipCPU, 2024-12-21:
Have you considered [this IP](https://github.com/ZipCPU/wbscope/blob/master/rtl/memscope.v) at all?  It's also designed to work like a scope--writing everything to AXI memory until you tell it to stop.  Then a second interface allows you to read back from the earliest memory to the most recent.  With a little bit of work, you could trim the "trigger" so that it only ever triggered on the last word in one of your sequences.

One difference from the concept you shared initially is that this one treats its memory as circular.  It will write and overwrite and overwrite again its memory until told to stop.  This allows you to stop on an event and see what lead up to the event.

Perhaps that might help?

closed 2025-01-01 by ZipCPU