ZipCPU/wb2axip #64: faxi_slave.v parse errors

pbozeman, 2024-12-31:
I just went through your blog entry on formal verification of axi cores and am working on a new one where I was planning to use this, but it looks like the version checked in was mid development. Is there some other repo or branch where I can find the version that corresponds to the blog documenting your results on your axi core?

There are a number of places where faxi_slave.v doesn't parse.  Most of them are issues like:

if (a && b && /* .... *) 

where that last && is just dangling. Example:

https://github.com/ZipCPU/wb2axip/blob/794f2028242519cb6c82b803a4bf3c107c8b4a90/bench/formal/faxi_slave.v#L1010C5-L1010C7

There is also a stray `endif at the end of the file.

Finally, while not a parse error, the setup for f_past_valid don't have a "reg f_past_valid", it just jumps straight to initializing it, so it bombs out with ERROR: Identifier `\f_past_valid' is implicitly declared and `default_nettype is set to none. Same with f_reset_length.

F_OPT_INITIAL is used, but isn't declared as a parameter.

next_rd_nbursts is used, but not declared anywhere. 

I was fixing things up to this point and then just stopped as this seems like it must be pretty far away from the version you wrote about on your blog.

Thanks!

ZipCPU, 2025-01-01:
The formal AXI properties found within this public github repo are only a subset of the actual properties I use to formally verify AXI designs.  They are not intended to be functional.  The actual properties are available to my Patreon sponsors.

To quote from the readme:

> ... There's also a (partial) formal property specification for an AXI (full) bus, but the one in the master branch is incomplete.  The complete set of AXI properties are maintained elsewhere.

Or to quote from the faxi_slave.v file itself:

> This file contains a subset of the formal properties which I've used to formally verify that a core truly follows the full AXI4 specification.

If conditions that appear as `(A && B && /* ... */)` are there to indicate that additional conditions are required, but not included within the property file itself.

Frankly, getting the AXI properties right took a lot of work.  It's not trivial.

If you would like a copy of the full property set, please support me on Patreon and then send me an e-mail.

Dan

closed 2025-01-01 by ZipCPU

pbozeman, 2025-01-01:
Ah, I didn't read the readme :)  I jumped there directly from the blog or google. I'll reach out over email and patreon.