ZipCPU/wb2axip #66: Vivado synthesis gated clock conversion is unable to pick up on AXIS2MM gated clock

mbiermann-mo-space, 2025-01-07:
Hi,
After struggling a lot with the reliability of Xilinx' DMAs I'm now in the process of switching to your axis2mm/ aximm2s IPs.

When using the gated clock option for axis2mm a critical warning is issued because the clock gating is using LUTs instead of onboard clocking resources.

According to [this article](https://adaptivesupport.amd.com/s/article/982650) vivado synthesis is able to convert gated clocks by using the synthesis option gated_clock_conversion and adding the attribute `gated_clock = "yes"` to the clock input, however the synthesis is unable to pick up on the usage of the gated clock.

I tried both with the attribute set to `OPT_CLKGATE ? "yes" : "no"` and to `"yes"` but neither option worked, so my guess is that the conversion tool is expecting the clock gating to work differently (Maybe it's because `gatep` and `gaten` are set on edges of the clock itself?).

ZipCPU, 2025-01-07:
Try setting `OPT_CLKGATE=1'b0`.  It's a simple one bit value.  There's no string comprehension within it, so I have no expectation that either `yes`, `no`, `true`, nor `false` will work.

Dan

mbiermann-mo-space, 2025-01-08:
To be honest I just saw the option and wanted to try it out, mainly out of curiosity but also because I wanted to add all necessary attributes to make the DMA integrate in Vivado as best as I could (I'd also like to add a PR for that later on).

In the article I mentioned (https://adaptivesupport.amd.com/s/article/982650) it is explained that Vivado synthesis should be able to detect clock gating and replace it with registers that have a clock enable port.
When the option is set and the right attribute is added to the clock port the clock gating conversion is run but it reports that no gated clocks are found.
I'm not familiar with how clock gating is done "usually" and as such have no idea how the synthesis expects it to look like, which is why I added the issue.

Using the clock gating is not necessary for my design so I just disabled the option, like you said.
As this is not an issue for me personally please feel free to just close the issue,
I mainly just wanted to see if you had some idea on how to do this 😄

closed 2025-01-10 by ZipCPU