ZipCPU/wb2axip #67: Current verilator fails on axispacker.v

j-f-d, 2025-01-09:
I'm new to verilog and FPGAs, but I tried to make in the checkout and it fails. I'm guessing that this might be a different version of verilator being more fussy?

```
$ verilator --version
Verilator 5.020 2024-01-01 rev (Debian 5.020-1)
$ make
tar --transform s,^,`date +%Y%m%d`-wb2axi/, -chjf `date +%Y%m%d`-wb2axi.tjz `find bench -name Makefile` `find bench -name "*.cpp"` `find bench -name "*.h"` `find rtl -name "*.v"` `find rtl -name Makefile` `find . -name "*.txt"` `find . -name "*.html"`
cd rtl ; make --no-print-directory
verilator -MMD -O3 -Wall -Wpedantic -cc axispacker.v
%Warning-SIDEEFFECT: axispacker.v:161:6: Expression side effect may be mishandled
                                       : ... Suggest use a temporary variable in place of this expression
  161 |    = pack( pre_tdata, pre_tstrb,
      |      ^~~~
                     ... For warning description see https://verilator.org/warn/SIDEEFFECT?v=5.020
                     ... Use "/* verilator lint_off SIDEEFFECT */" and lint_on around source to disable this message.
%Warning-SIDEEFFECT: axispacker.v:160:25: Expression side effect may be mishandled
                                        : ... Suggest use a temporary variable in place of this expression
  160 |   { pck_tdata, pck_tstrb, pck_tkeep }
      |                         ^
%Error: Exiting due to 2 warning(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
make[1]: *** [Makefile:170: obj_dir/Vaxispacker.mk] Error 1
make[1]: *** Deleting file 'obj_dir/Vaxispacker.mk'
make[1]: *** [obj_dir/Vaxispacker.mk] Deleting file 'obj_dir/Vaxispacker.h'
make[1]: *** [obj_dir/Vaxispacker.mk] Deleting file 'obj_dir/Vaxispacker.cpp'
make: *** [Makefile:56: verilated] Error 2
```

ZipCPU, 2025-01-09:
Thank you for bringing this to my attention.  I now have a tested fix that I should be able to post shortly.

ZipCPU, 2025-01-10:
Fixed in 64675ca7b0d27df6ba4d0206645a76951535bab4

closed 2025-01-10 by ZipCPU