ZipCPU/wb2axip #69: axis2mm sometimes hangs when r_continuous is set to 1

mbiermann-mo-space, 2025-01-14:
Maybe I'm misinterpreting the correct usage of r_continuous.
The FIFO being reset means that some data is lost correct?
I'd like to avoid that however this isn't critical for my application so for now a workaround is to just not use r_continuous.

The core reports it is busy however S_AXIS_TREADY stays at 0 and the transfer is never finished (even after several seconds r_busy stays 1 for small transfer sizes).

For me it happens when I:
- direct the core to write 1MiB
- don't start another transfer immediately after (no error bit is set though)
- after a while start another transfer
  - from a new address instead of the one where the core left off (the address is the same as for the first transfer)
  - with a comparatively low length (16 bytes and 1024 bytes both don't work, 4096 bytes does)

ZipCPU, 2025-01-14:
There's more to this story.

- `S_AXIS_TREADY` will only be low if the FIFO is full.
- The FIFO can only be emptied when the IP is programmed, or if it is idle and `r_continuous` is clear.
- The FIFO can fill if `r_continuous` is set--even before the core is programmed for a transfer.  That's the purpose of `r_continuous`.
- This particular IP doesn't stop on `S_AXIS_TLAST`.  Rather, it stops when the given amount of data has been transferred.  I've been working on another S2MM that instead stops on TLAST, but ... that's not this one.

If this doesn't help, then I'll need you to do some digging for me to discover why `S_AXIS_TREADY` (i.e. `sskd_ready`) is being held low.

mbiermann-mo-space, 2025-01-16:
That sounds like the FIFO is full which makes sense with `r_continuous` set and no transfer being started for a while.

`S_AXIS_TLAST` shouldn't be an issue, my application doesn't depend on it at all since I don't transfer packets.

The issue is really more that the core doesn't seem to start transferring the data from the FIFO to `M_AXI` and as such doesn't finish the transaction.

After digging around a bunch I'm pretty sure I found the issue:
```verilog
	always @(*)
	if (|aw_requests_remaining[LGLENW-1:LGMAXBURST])
		sufficiently_filled = |data_available[LGFIFO:LGMAXBURST];
	else
		sufficiently_filled = (data_available[LGMAXBURST-1:0]
				>= aw_requests_remaining[LGMAXBURST-1:0]);
```

When `aw_requests_remaining` is less than `1 << LGMAXBURST` the whole of `data_available` should be checked instead of just `data_available[LGMAXBURST-1:0]`, in my case `data_available` is `0x200` so `data_available[LGMAXBURST-1:0]` is 0.

I'll create a PR.

ZipCPU, 2025-01-17:
The fix you are recommending is not nearly as straightforward as it looks.  Before I dive into this fix, I'd like to first be able to simulate it.  My plan is to simulate it as part of the [axidmatest](https://github.com/ZipCPU/axidmatest) repository.

Can you share or suggest any particular setup or target script that would trigger this bug?

Dan

mbiermann-mo-space, 2025-01-20:
As far as I understand it, the bug is triggered when the FIFO is full and `aw_requests_remaining` is smaller than the maximum burst - so when only a single request is remaining.

The way I triggered it was by setting `r_continuous` to 1, waiting for the FIFO to fill up completely (so all bits in `data_available[LGMAXBURST-1:0]` are 0) and then starting a transaction that's smaller than the maximum burst size.

I hope this information is enough, I don't have access to the axidmatest repository so I don't know what kind of script you need.

Max

ZipCPU, 2025-01-20:
Sorry, I mis-typed the name of the testing distribution.  It should be [axidmacheck](https://github.com/zipcpu/axidmacheck).  Yes, you should have access to it--I've historically never paid github for private repositories.

Let me see how far I can get with the description you've just given me.

ZipCPU, 2025-01-22:
I can now reproduce your bug.

Your PR looks quite reasonable given what's going on.  Thank you for submitting it.

Let me double check with the formal tools that this doesn't bust anything.  Otherwise, this looks good.

Dan

mbiermann-mo-space, 2025-01-23:
Great! I'm happy to give back to this project 👍 

Max

closed 2025-01-23 by ZipCPU

ZipCPU, 2025-01-23:
Merged.  Lint waiver added.  Again, thank you.

Dan