ZipCPU/wb2axip #72: Skidbuffer asserts o_ready while being reset

mbiermann-mo-space, 2025-01-31:
I was wondering why `S_AXIS_TREADY` of axis2mm was held high without the fifo being filled and found that the skidbuffer holds `o_ready` high as long as `i_reset` is asserted.

This results in axis2mm `S_AXIS_TREADY` being 1 when idle without the option `OPT_TREADY_WHILE_IDLE` being used.

The code in question is:

```verilog
initial if (OPT_INITIAL) r_valid = 0;
always @(posedge i_clk)
if (i_reset)
	r_valid <= 0;
else if ((i_valid && o_ready) && (o_valid && !i_ready))
	// We have incoming data, but the output is stalled
	r_valid <= 1;
else if (i_ready)
	r_valid <= 0;
```
and
```verilog
assign o_ready = !r_valid;
```
Maybe I have a misunderstanding here but my expectation was that the skidbuffer only asserts `o_ready` when the data from the stream is actually being consumed.

mbiermann-mo-space, 2025-01-31:
I tried changing the second code snippet to `assign o_ready = !r_valid && !i_reset;`, in that case however axis2mm always reports overflow errors.

My guess for the reason is that `!sskd_valid` is used in overflow detection - is that correct? Intuitively I would've though the check would be `!S_AXIS_TVALID` since that's the signal that's used in detecting stalling (`last_stalled <= (!i_reset) && (S_AXIS_TVALID && !S_AXIS_TREADY)`).

ZipCPU, 2025-01-31:
1. `sskd_valid` is used in overflow detection.  You are correct, this is an error.  `S_AXIS_TVALID` should be used instead.  The problem is that the skidbuffer assumes a well behaved AXI Stream source, and such a well behaved source is not able to overflow anything.  Hence the detection should be made on the actual AXI stream (which is unable to follow the stream protocol).  ("Unable to follow the stream protocol" is actually pretty common with digitizer sources, but that's another story ... this IP is supposed to be able to detect and handle that properly.)
2. The skid buffer should not be changed.  The skid buffer is properly handling the reset.  This is appropriate for the system wide reset which it was designed to handle, although perhaps not nearly as appropriate for a local reset as in the environment of the MM2S controller.

Now, with all that said, I'm ... struggling a bit to understand the context where this is a problem.  If `i_reset` is set, then everything must be reset--to include the bus.  Resetting the DMA without resetting the bus will have nasty consequences.  So we must then be discussing a case where `!r_busy && (!r_continuous || r_err)`, right?  Hence, we're discussing a case between transactions (`!r_busy`), and either the continuous bit is not set, or if it is set an error has occurred.  In the case of an error, all bets are off.  So are we discussing what happens to data in this IP before the DMA is programmed for the first time to set `r_continuous`?

Dan

mbiermann-mo-space, 2025-01-31:
Hi Dan,

I encountered this with `r_continuous` being unset (I was trying various things because the data that was written didn't match the stream input - but I found that the written data also doesn't match the memory mapped bus contents so the issue has to be somewhere else - although all that comes after is a Xilinx Smartconnect and my Zynq PS...).

In the process I found that `S_AXIS_TREADY` was set before my first interaction with the DMA, which was why I started looking at the skidbuffer.

The skidbuffers `i_reset` is wired to `reset_fifo` which means with `r_continuous` not being set `S_AXIS_TREADY` is also always set between transactions even when `OPT_TREADY_WHILE_IDLE` is off. I'm unsure but maybe another way would be to only reset the skidbuffer when resetting the whole DMA?

Max

ZipCPU, 2025-01-31:
So ... your problem is that `S_AXIS_TREADY` is set for a clock cycle prior to the IP being set up to collect data, even though `OPT_TREADY_WHILE_IDLE` is zero.  Yes?

... and you are proposing something like:

`reset_fifo = i_reset || (!OPT_TREADY_WHILE_IDLE && !r_busy && (!r_continuous || r_err)` ?

That doesn't seem right though.  At issue is the fact that the FIFO is being reset, in addition to the skidbuffer.  The FIFO _needs_ to be reset between captures if there's no desire to be continuous--otherwise you'll end up with discontinuous data during a collect.  Even in the case of continuous data, the FIFO (and skid buffer) needs to be reset on overflow, or (again) you'll get discontinuous data during a collect.

Perhaps a better approach would be passing a reset to the skidbuffer of `(OPT_TREADY_WHILE_IDLE ? i_reset : reset_fifo)`?  No, this won't work either, since it might (again) lead to discontinuous data.

Would a better option be returning a temporary signal `skidbuffer_ready` intsead of `S_AXIS_TREADY`, and then setting `S_AXIS_TREADY = skidbuffer_ready && (OPT_TREADY_WHILE_IDLE || !reset_fifo)`?  This will impact timing somewhat and may even defeat the purpose of the skidbuffer in the first place, but might accomplish what you are looking for ... perhaps.  (I'm thinking "out loud" here ...)

mbiermann-mo-space, 2025-02-04:
I think so too, other than the initial and reset value there shouldn't be any changes necessary.
Do you want me to create a PR?

ZipCPU, 2025-02-04:
Sure, but do try it first and let me know how well it works for you, then I'll accept a PR.

mbiermann-mo-space, 2025-02-05:
I'm now done trying it out and it works as expected, all contents of the stream get written to memory and the DMA stalls the stream when it isn't ready 👍

I'll create a PR

ZipCPU, 2025-02-06:
God bless, be well.