ZipCPU/wb2axip #74: AXISPACKER generating extreme timing paths

KireinaHoro, 2025-02-26:
I'm trying to integrate the AXISPACKER IP core into my design, after a couple stages of header extraction modules that I wrote myself.  I instantiated the IP core for a bus width of 512b on Xilinx UltraScale+, clocked at 250 MHz.

I'm observing extremely long combinatorial paths in the core of a whopping 56 stages:
[image]

```
Max Delay Paths
--------------------------------------------------------------------------------------
Slack (VIOLATED) :        -16.734ns  (required time - arrival time)
  Source:                 i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.REG_OUTPUT.ro_valid_reg_rep__0/C
                            (rising edge-triggered cell FDRE clocked by clk_out1_design_1_clk_wiz_0_0  {rise@0.000ns fall@2.000ns period=4.000ns})
  Destination:            i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/beat_skd/LOGIC.REG_OUTPUT.o_data_reg[418]/D
                            (rising edge-triggered cell FDRE clocked by clk_out1_design_1_clk_wiz_0_0  {rise@0.000ns fall@2.000ns period=4.000ns})
  Path Group:             clk_out1_design_1_clk_wiz_0_0
  Path Type:              Setup (Max at Slow Process Corner)
  Requirement:            4.000ns  (clk_out1_design_1_clk_wiz_0_0 rise@4.000ns - clk_out1_design_1_clk_wiz_0_0 rise@0.000ns)
  Data Path Delay:        21.141ns  (logic 5.491ns (25.974%)  route 15.650ns (74.026%))
  Logic Levels:           56  (LUT4=10 LUT5=4 LUT6=42)
  Clock Path Skew:        0.443ns (DCD - SCD + CPR)
    Destination Clock Delay (DCD):    6.664ns = ( 10.664 - 4.000 ) 
    Source Clock Delay      (SCD):    6.061ns
    Clock Pessimism Removal (CPR):    -0.160ns
  Clock Uncertainty:      0.060ns  ((TSJ^2 + DJ^2)^1/2) / 2 + PE
    Total System Jitter     (TSJ):    0.071ns
    Discrete Jitter          (DJ):    0.097ns
    Phase Error              (PE):    0.000ns
  Clock Net Delay (Source):      2.794ns (routing 1.333ns, distribution 1.461ns)
  Clock Net Delay (Destination): 3.069ns (routing 1.211ns, distribution 1.858ns)

    Location             Delay type                Incr(ns)  Path(ns)    Partition      PBlock               Netlist Resource(s)
  -------------------------------------------------------------------    -------------------------------------------------------
                         (clock clk_out1_design_1_clk_wiz_0_0 rise edge)
                                                      0.000     0.000 r                                      
    AY26                                              0.000     0.000 r  reconfigurable                      F_PRGC0_CLK_P (IN)
                         net (fo=0)                   0.100     0.100    reconfigurable                      i_app/i_eci_gateway/i_prgc0/I
    HPIOBDIFFINBUF_X1Y154
                         DIFFINBUF (Prop_DIFFINBUF_HPIOBDIFFINBUF_DIFF_IN_P_O)
                                                      0.474     0.574 r  reconfigurable pblock_dynamic       i_app/i_eci_gateway/i_prgc0/DIFFINBUF_INST/O
                         net (fo=1, routed)           0.050     0.624    reconfigurable                      i_app/i_eci_gateway/i_prgc0/OUT
    AY26                 IBUFCTRL (Prop_IBUFCTRL_HPIOB_M_I_O)
                                                      0.000     0.624 r  reconfigurable pblock_dynamic       i_app/i_eci_gateway/i_prgc0/IBUFCTRL_INST/O
                         net (fo=1, routed)           0.335     0.959    reconfigurable                      i_app/i_eci_gateway/i_prgc0_n_0
    BUFGCE_X1Y148        BUFGCE (Prop_BUFCE_BUFGCE_I_O)
                                                      0.028     0.987 r  reconfigurable pblock_dynamic       i_app/i_eci_gateway/i_clk_io_bufg/O
                         net (fo=25736, routed)       2.140     3.127    reconfigurable                      i_app/design_1_i/clk_wiz_0/inst/clk_in1
    MMCM_X1Y6            MMCME4_ADV (Prop_MMCM_CLKIN1_CLKOUT0)
                                                     -0.127     3.000 r  reconfigurable pblock_dynamic       i_app/design_1_i/clk_wiz_0/inst/mmcme4_adv_inst/CLKOUT0
                         net (fo=1, routed)           0.239     3.239    reconfigurable                      i_app/design_1_i/clk_wiz_0/inst/clk_out1_design_1_clk_wiz_0_0
    BUFGCE_X1Y149        BUFGCE (Prop_BUFCE_BUFGCE_I_O)
                                                      0.028     3.267 r  reconfigurable pblock_dynamic       i_app/design_1_i/clk_wiz_0/inst/clkout1_buf/O
    X2Y9 (CLOCK_ROOT)    net (fo=213082, routed)      2.794     6.061    reconfigurable                      i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/app_clk
    SLICE_X73Y352        FDRE                                         r  reconfigurable pblock_dynamic       i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.REG_OUTPUT.ro_valid_reg_rep__0/C
  -------------------------------------------------------------------    -------------------------------------------------------
    SLICE_X73Y352        FDRE (Prop_DFF_SLICEL_C_Q)
                                                      0.070     6.131 r  reconfigurable pblock_dynamic       i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.REG_OUTPUT.ro_valid_reg_rep__0/Q
                         net (fo=4, routed)           0.119     6.250    reconfigurable                      i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.REG_OUTPUT.ro_valid_reg_rep__0_n_0
    SLICE_X73Y352        LUT6 (Prop_F6LUT_SLICEL_I5_O)
                                                      0.133     6.383 f  reconfigurable pblock_dynamic       i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.r_data[639]_i_2126/O
                         net (fo=10, routed)          0.101     6.484    reconfigurable                      i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.r_data[639]_i_2126_n_0
    SLICE_X73Y351        LUT6 (Prop_E6LUT_SLICEL_I2_O)
                                                      0.047     6.531 f  reconfigurable pblock_dynamic       i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.r_data[639]_i_2130/O
                         net (fo=8, routed)           0.251     6.781    reconfigurable                      i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.r_data[639]_i_2130_n_0
    SLICE_X73Y350        LUT5 (Prop_F6LUT_SLICEL_I3_O)
                                                      0.031     6.812 r  reconfigurable pblock_dynamic       i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.r_data[639]_i_2605/O
                         net (fo=22, routed)          0.193     7.006    reconfigurable                      i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.r_data[639]_i_2605_n_0
    SLICE_X73Y348        LUT6 (Prop_D6LUT_SLICEL_I0_O)
                                                      0.089     7.095 f  reconfigurable pblock_dynamic       i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.r_data[639]_i_2443/O
                         net (fo=13, routed)          0.319     7.414    reconfigurable                      i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.r_data[639]_i_2443_n_0
    SLICE_X74Y347        LUT6 (Prop_E6LUT_SLICEM_I2_O)
                                                      0.088     7.502 f  reconfigurable pblock_dynamic       i_app/NicEngine_inst/AxiDmaPlugin_logic_packer/slave_skd/LOGIC.r_data[639]_i_2310/O
[... 169 lines cut ...]
                                                      0.024     4.858 r  reconfigurable pblock_dynamic       i_app/i_eci_gateway/i_clk_io_bufg/O
                         net (fo=25736, routed)       1.943     6.801    reconfigurable                      i_app/design_1_i/clk_wiz_0/inst/clk_in1
    MMCM_X1Y6            MMCME4_ADV (Prop_MMCM_CLKIN1_CLKOUT0)
                                                      0.560     7.361 r  reconfigurable pblock_dynamic       i_app/design_1_i/clk_wiz_0/inst/mmcme4_adv_inst/CLKOUT0
                         net (fo=1, routed)           0.210     7.571    reconfigurable                      i_app/design_1_i/clk_wiz_0/inst/clk_out1_design_1_clk_wiz_0_0
    BUFGCE_X1Y149        BUFGCE (Prop_BUFCE_BUFGCE_I_O)
                                                      0.024     7.595 r  reconfigurable pblock_dynamic       i_app/design_1_i/clk_wiz_0/inst/clkout1_buf/O
[... cut ...]
                         required time                         10.467                                            
                         arrival time                         -27.202                                            
  -------------------------------------------------------------------
                         slack                                -16.734                                            

```

LUT usage of the core is also (somewhat) surprising:
[image]

I'm not entirely sure what is causing this, and if there's a way around it.  Can you take a look?  Thanks a lot :)

ZipCPU, 2025-02-26:
Yeah, sure ... let me take a look ...

ZipCPU, 2025-02-26:
I'm showing a maximum logic depth of 9 for a data width of 32.  How big is your data width?

KireinaHoro, 2025-02-26:
> I'm showing a maximum logic depth of 9 for a data width of 32. How big is your data width?

It's 512 bits -- I'm aware that this might expand the depth since it is in principle a fully connected barrel shifter, but not sure how much.

ZipCPU, 2025-02-27:
Hmm ... is this a time sensitive task?  Can you afford latency?  You might need an application specific solution.  I don't know.  Yes, the current axispacker algorithm logic will bloom with a wide data width.

For the cost of (about) (512/8) or 64 (pipeline) clocks, you can adjust the logic so it first pushes everything to the right (towards the LSB), 8b at a time.  A second stage could then merge words together using the full barrel shifter.

Can you share anything about the data you are working with that requires something like this?

KireinaHoro, 2025-02-27:
> Can you share anything about the data you are working with that requires something like this?

I'm using the core inside a NIC that I'm building.  I have a module to strip a fixed number of bytes off the start of the AXI-Stream (configurable at elaboration).  The module strips the header by setting TKEEP to zero for the removed bytes, as I explained earlier.

I chain these modules to remove Ethernet headers, IP headers, UDP/TCP headers, etc.  Eventually I end up with an AXI stream that has some number of NULL bytes at the beginning -- `len(sum(headers)) % 64` to be exact.  Since the packet might have travelled through different paths in the decoder pipeline, e.g. Ethernet - IP - TCP vs. Ethernet - IPv6 - UDP, the number of null bytes are not always fixed.

I think a design that splits into multiple stages and then merges them back sounds reasonable -- 8b at a time sounds a bit extreme, maybe 32b * 16 would still work fine?  We could probably stick more logic into one stage (and still hit 250 MHz).  If the design would be fully pipelined, 16 cycles of latency (~64ns) is not too much of an issue.

ZipCPU, 2025-02-28:
Sounds to me like the "answer" to the AXISPACKER would be a bit of an algorithm rewrite, so that it packs data in stages.

As for network packet handling, have you considered using a (subtly) different protocol, like the [AXI Network protocol](https://github.com/ZipCPU/eth10g/blob/master/doc/axin.pdf)?  It would have a logic advantage over the AXI Stream protocol you are using, simply because it doesn't use the TKEEP but rather a BYTES feature.  You can then dump your packet into [something like a more conventional gearbox](https://github.com/zipcpu/zipcpu/blob/master/rtl/zipdma/zipdma_rxgears.v) to pack things back together to the full bus width.